bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #2: Board Schematic

thisiskeithb, 2019-05-11:
Do you have a board schematic for the SKR Mini 1.1?

Andypos, 2019-05-24:
I need schematic for USB and USART ( what usart you use for LCD 3.5?)

IvanLev, 2019-05-25:
> I need schematic for USB and USART ( what usart you use for LCD 3.5?)

Usb is most probably a direct connection to the MCU so no need for the schmatic(pins Pa11 and PA12) and just probed my board with a multimeter the LCD connector is connected to USART1. TX- PA9, RX- PA10.

Andypos, 2019-05-26:
> > I need schematic for USB and USART ( what usart you use for LCD 3.5?)
> 
> Usb is most probably a direct connection to the MCU so no need for the schmatic(pins Pa11 and PA12) and just probed my board with a multimeter the LCD connector is connected to USART1. TX- PA9, RX- PA10.

Ok. Thx. I need more info about pullup ,pulldown resistors and current in power 5V  for connect   small servo .

thisiskeithb, 2019-08-24:
Closing since schematics [were finally uploaded](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/blob/8196f18f13b61a14df8d55acdba32b5e7f12421b/hardware/SKR-mini-V1.1sch.pdf).

closed 2019-08-24 by thisiskeithb