bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #3: unable to compile

Udo-Nbg, 2019-05-13:
stm32f103RMarlin-bugfix-2.0V20190126-2004P-16div
stm32f103RMarlin-bugfix-2.0V20190126-2004P-128div

both Versions are not to compile.
also the originial Version without changes.

WayeeCool, 2019-05-14:
I had the same issue and [opened an issue about it](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/issues/1).  It's been a week, still no response and no update from bigtreetech.  I was able to figure out the error preventing compile and correct it but the firmware is far from finished and I am surprised that bigtreetech is actually shipping a product without firmware support.

To fix the issue `#include "spi.h"` needs to be changed to `#include "SPI.h"` on `line 40 of Marlin/src/sd/Sd2Card.cpp`. The PlatformIO compiler seems to be case-sensitive and fixing this capitalization issue results in a successful compile with no critical errors.

Udo-Nbg, 2019-05-14:
Thank You but this DID **NOT SOLVE** THE PROBLEM

Environment BIGTREE_SKR_MINI            [ERROR]

jmdearras, 2019-05-19:
I get many "duplicate inline statements".

Udo-Nbg, 2019-05-24:
SOLVED

look at this: 
https://github.com/MarlinFirmware/Marlin/issues/14053?fbclid=IwAR3EkbqhnLRONBdWv-P7qqvf3UwBPdfdNoptKzOukwBU0lwfK4EUIO6gTgk

https://github.com/MarlinFirmware/Marlin/issues/14053?fbclid=IwAR3EkbqhnLRONBdWv-P7qqvf3UwBPdfdNoptKzOukwBU0lwfK4EUIO6gTgk#issuecomment-495023530

closed 2019-05-24 by Udo-Nbg

Udo-Nbg, 2019-08-04:
Marlin-bugfix-2.0.x branch 4350 ist the Solve.
It will compile successfully.