bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #4: Compile issue

Foxbiker, 2019-05-24:
stm32f103RMarlin-bugfix-2.0V20190126-2004P-16div
stm32f103RMarlin-bugfix-2.0V20190126-2004P-128div

both Versions are not to compile. 

This solution not help:
https://github.com/MarlinFirmware/Marlin/issues/14053

Nothing help

Processing BIGTREE_SKR_MINI (platform: ststm32@<4.4.0; board: genericSTM32F103RC; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz 48KB RAM (256KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.3.4
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD> #c8ac22f
|-- <SailfishRGB_LED> #2426fa2
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> #3a18be5
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL.cpp.o
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp.o
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp.o
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_timers_STM32F1.cpp.o
In file included from c:\users\lisak\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:19,
from c:\users\lisak\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\cstdlib:75,
from c:\users\lisak\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdlib.h:36,
from C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:41,
from C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
from Marlin\src\HAL\HAL_STM32F1\../shared/Marduino.h:33,
from Marlin\src\HAL\HAL_STM32F1\HAL.h:40,
from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:34:
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\system\libmaple/stm32f1/include/series/gpio.h:487:15: error: duplicate 'inline'
static inline __always_inline void afio_exti_select(exti_num exti, exti_cfg port) {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\system\libmaple\include/libmaple/nvic.h:112:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_enable() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\system\libmaple\include/libmaple/nvic.h:119:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_disable() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
static inline __always_inline void interrupts() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~
In file included from c:\users\lisak\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:19,
from c:\users\lisak\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\cstdlib:75,
from c:\users\lisak\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdlib.h:36,
from C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/wirish.h:41,
from C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Arduino.h:30,
[... 46 lines cut ...]
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
static inline __always_inline void interrupts() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
static inline __always_inline void interrupts() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
static inline __always_inline void interrupts() {
^~~~~~~~~~~~~~~
C:\users\lisak\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp.o] Error 1
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_timers_STM32F1.cpp.o] Error 1
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp.o] Error 1
======================================================= [ERROR] Took 5.39 seconds =======================================================

Mobile-Dom, 2019-05-25:
having this same issue, board is currently useless to me if firmware cannot be compiled.

ojbl, 2019-05-27:
Try running this command in the platformio terminal
"platformio run -e STM32F1"
and then compile.

Foxbiker, 2019-06-02:
I try run this command in the platformio terminal
"platformio run -e STM32F1"
but without success.

[image]

rhialto56, 2019-06-09:
I compile after many tries... Ans need to use u8glib_hal ans do some modifications in code