bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #5: Marlin 2.0 with BLTouch Compiling issue

anm8tor, 2019-06-05:
So I was in the process of changing some settings to the Marlin 2.0 build provided here and ran into several compiling errors. I'm not sure I am posting the correct data for someone to help, but this is my first time ever posting. So here goes; I also included a screenshot of the "problems" portion from PIO:

In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41,
from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:957:8: error: #error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."
#error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."
^~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1777:4: error: #error "TMC2208 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
#error "TMC2208 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
^~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1841:4: error: #error "TMC2208 Software Serial is supported only on AVR and LPC1768 platforms."
#error "TMC2208 Software Serial is supported only on AVR and LPC1768 platforms."
^~~~~
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:36,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\I2cEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/Conditionals_LCD.h:51: warning: "ENCODER_PULSES_PER_STEP" redefined
#define ENCODER_PULSES_PER_STEP 1

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:35,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\I2cEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/../../Configuration.h:1547: note: this is the location of the previous definition
#define ENCODER_PULSES_PER_STEP 2

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:36,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\shared\servo.cpp.o

from Marlin\src\HAL\shared\I2cEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/Conditionals_LCD.h:52: warning: "ENCODER_STEPS_PER_MENU_ITEM" redefined
#define ENCODER_STEPS_PER_MENU_ITEM 1

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:35,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\I2cEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/../../Configuration.h:1553: note: this is the location of the previous definition
#define ENCODER_STEPS_PER_MENU_ITEM 2

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:36,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\SpiEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/Conditionals_LCD.h:51: warning: "ENCODER_PULSES_PER_STEP" redefined
#define ENCODER_PULSES_PER_STEP 1

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:35,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\SpiEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/../../Configuration.h:1547: note: this is the location of the previous definition
#define ENCODER_PULSES_PER_STEP 2

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:36,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\SpiEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/Conditionals_LCD.h:52: warning: "ENCODER_STEPS_PER_MENU_ITEM" redefined
#define ENCODER_STEPS_PER_MENU_ITEM 1

In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:35,
from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\shared\SpiEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/../../Configuration.h:1553: note: this is the location of the previous definition
#define ENCODER_STEPS_PER_MENU_ITEM 2

In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41,
from Marlin\src\HAL\HAL_STM32F1\persistent_store_flash.cpp:32:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:957:8: error: #error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."
#error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."
^~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1777:4: error: #error "TMC2208 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
#error "TMC2208 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
^~~~~
[... 327 lines cut ...]
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:164:13: warning: unnecessary parentheses in declaration of '_spi1_this' [-Wparentheses]
static void (*_spi1_this);
^
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:165:13: warning: unnecessary parentheses in declaration of '_spi2_this' [-Wparentheses]
static void (*_spi2_this);
^
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:167:13: warning: unnecessary parentheses in declaration of '_spi3_this' [-Wparentheses]
static void (*_spi3_this);
^
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:167:15: warning: '_spi3_this' defined but not used [-Wunused-variable]
static void (*_spi3_this);
^~~~~~~~~~
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:165:15: warning: '_spi2_this' defined but not used [-Wunused-variable]
static void (*_spi2_this);
^~~~~~~~~~
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:164:15: warning: '_spi1_this' defined but not used [-Wunused-variable]
static void (*_spi1_this);
^~~~~~~~~~
C:\users\anm8t\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src/SPI.h:163:16: warning: 'ff' defined but not used [-Wunused-variable]
static uint8_t ff = 0XFF;
^~
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\Marlin.cpp.o] Error 1
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\core\utility.cpp.o] Error 1

[image]

anm8tor, 2019-06-05:
Ignore the BLTOUCH errors, I was able to get those resolved by adding "#define SERVO0_PIN     PA2" to the pins_BIGTREE_SKR_MINI_V1.1.h file

rhialto56, 2019-06-09:
You must define all pins in your pin_board.h

anm8tor, 2019-06-09:
Yes,  I understand. Are you able to provide any help on which pins are defined as what?

closed 2019-06-12 by anm8tor