bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #7: Accidently corrupted the bootloader

grblmpfh, 2019-06-17:
Hi,
I have accendently corrupted the bootloader and cannot find the source or a .bin file. Could you please publish the bootloader too

THX
Chris

3DSmitty, 2019-06-20:
Yes I could use this as well, please?

1kemanciii, 2019-06-22:
yes. I have the same problem.

aetbruce, 2019-06-27:
Please respond, it's been nearly two weeks since issue opened.

ThWoell, 2019-06-30:
Count me in,
updating via sd-card only sucks...

3DSmitty, 2019-06-30:
Here is an alternative solution. You can still install Marlin without the bootloader if you have a stlink.

1. Download and install the STM32 ST-LINK Utility (found on ST website)
2. Connect stlink to ICSP (Use 3.3V to connect to 5V)
3. In platformio.ini comment out this line:
       extra_scripts = buildroot/share/PlatformIO/scripts/STM32F1_SKR_MINI.py
   to:
       # extra_scripts = buildroot/share/PlatformIO/scripts/STM32F1_SKR_MINI.py
4. Build and use  STM32 ST-LINK Utility to upload firmware.bin to board

Hope that helps.

1kemanciii, 2019-06-30:
@3DSmitty Can you tell with video pls

ThWoell, 2019-06-30:
if you own a st-link, try the generic out of them:

https://github.com/zoobab/STM32duino-bootloader

it is a fork of the original with longer waiting time (1o sec.) for begin of dfu-upload

1kemanciii, 2019-06-30:
no st-link :(

ThWoell, 2019-06-30:
It seems there is a Bootloader installed, in the Arduino IDE (on MAC OS-X) the board will be identified as maple-mini after installing the stm32 tools from rogerclarkmelbourne...
since the maple mini sports some extra components for the USB-interface which are needed to switch to dfu mode, it ain't possible to do so if those are missing.

should also work with ftdi serial adapter, but you have to pull the boot0 pin of the stm to ground...

grblmpfh, 2019-06-30:
@ThWoell's linked bootloader will not work without changes either in the bootloader or in marlin. Marlin is compiled to run at an offset of 0x7000 (28k) and the bootloader looks at offsets of 0x2000 or 0x5000. S this has to be fixed first. It should be possible to compile marlin without a bootloader offset (so 0x00000) and use the serial/uart upload that is integrated in every STM32.

grblmpfh, 2019-06-30:
In PlatformIO/Marlin look at ...Marlin//buildroot/share/PlatformIO/ldscripts/STM32F1_SKR_MINI.ld to change the bootloader offset.

1kemanciii, 2019-06-30:
do you have a working assembly?

grblmpfh, 2019-06-30:
Unfortunatly not... haven't had the time to work on it.

1kemanciii, 2019-06-30:
can you share when you do

grblmpfh, 2019-06-30:
of course :)

grblmpfh, 2019-06-30:
I'm too old for this kind of work... I cant find a spot where to get the BOOT0 signal exept for pin 60 of the STM32. But for me that is far to small to see or to solder something to it to access that signal... 

To flash a new bootloader or new software with an ftdi serial adapter you have to pull that pin high while powering up. That would enable the built in serial bootloader. 
Maybe someone else has an idea...

3DSmitty, 2019-06-30:
That's the problem the BOOT0 is hardwired to always be in programming mode that's why you connect 3.3V instead of 5 with the stlink. You can always read/write to the ISCP port. with reboots of course.

3DSmitty, 2019-06-30:
@ThWoell that bootloader does work but because BOOT0 is hardwired it is always in DFU mode.

3DSmitty, 2019-06-30:
Guys, honestly you don't really even need a bootloader if you have a stlink. Marlin runs just fine without a bootloader and uploaded via stlink. We have done a ton of development on this board in the main Marlin repo, fyi

1kemanciii, 2019-06-30:
how to stlink loaded

1kemanciii, 2019-06-30:
how to vscode send

3DSmitty, 2019-06-30:
@1kemanciii there are plenty of youtube videos on that

1kemanciii, 2019-06-30:
I couldnt find it. can you link

3DSmitty, 2019-06-30:
How about this one
https://www.youtube.com/watch?v=gBcdHTUzRUU

ThWoell, 2019-07-01:
@3DSmitty: Sorry I have to ask again (being new to the world of STM32): You wrote the bootloader works always in DFU mode, but on which interface? USB, Serial, JTAG?
What quite works is uploading Firmware by SD-Card. Since I wanted to use the board as a replacement in an existing printer it would require a good amount of machining to put a slot for the sd-card reader into the metal housing which I would like to avoid.
Tried the serial connection yesterday but without success, maybe it was too late in the evening? It would be easy to fiddle a pair of wires through the holes in the housing and connect it to a serial adapter if you can state that this solution will work. (Or an st-link of course, if that would be the way)

3DSmitty, 2019-07-01:
@ThWoell I tried the generic_boot20_pa1.bin from https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/binaries . It does work but the USB is always in DFU mode. I guess you could use it if you connected a second serial to USB adapter to the TFT interface (which is the second serial port) but I have not confirmed the second serial port works with this method. If you wiped out your original bootloader though you are still going to need a stlink to even get the bootloader uploaded and at that point you might as well just bypass the bootloader completely. Hope that makes sense.

ThWoell, 2019-07-01:
Just to clarify: 
My Board is "as delivered" with the original bootloader. It is possible to communicate with the Printer over the USB-Interface e.g. with Pronterface or the Arduino. I can exchange Firmware via micro-SD.  I´m not able to upload Firmware over USB (which was the way on Arduino-based 8-bit boards) 
If I would exchange the "original" Bootloader with the generic it will give me the other way, upload via USB by DFU but no USB communication with e.g. pronterface. 
So I will forget about exchanging Bootloader and leave the board uninstalled in the housing as long as everything ist tweaked correctly in Marlin for my machine.

Thanks for Helping ot!

1kemanciii, 2019-07-14:
where can i find the original bootloader?

ThWoell, 2019-07-14:
At the moment nowhere but on your board...
Since the opening of this thread, no one from Bigtreetech jumped in.

Bansh1990, 2019-07-29:
try this
flash from The address is 0x08000000

[SKR-E3-SIP-bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/files/3440943/SKR-E3-SIP-bootloader.zip)

dinamitemic, 2019-11-05:
I'm in the same situation. Tryied to upload fw from st-link, then figured out that I've lost my bootloader since the marlin fw had not to start from 0x08000000 when uploaded directly.
@Bansh1990 I've also tryied to do what you suggested, started the upload then immediately stopped and since that moment I can't even reconnect to the micro through st-link.
So what can I do now? Nothing? Just take my board and put it in the rubbish!!!
@bigtreetech

dinamitemic, 2019-11-05:
Ok, after digging a bit more around I've found everithing that I was looking for.
Like it is mentioned [here](https://github.com/MarlinFirmware/Marlin/pull/14041#issuecomment-513607940) the right addresses in order to program it from stlink are:
-0x08000000 for the bootloader
-0x08007000 for the firmware.bin compiled in platformio.
[Here](https://github.com/MarlinFirmware/Marlin/pull/14041#issuecomment-513608547) you can also find the stock bootloader.
Hope to help someone else like me loosing hour in this bad documented issues.

davidcgu, 2020-01-02:
> Ok, after digging a bit more around I've found everithing that I was looking for.
> Like it is mentioned [here](https://github.com/MarlinFirmware/Marlin/pull/14041#issuecomment-513607940) the right [...]

Hello Mate,

First of all I wish you a happy new 2020!!!!!!
Thanks for your add, I see on the link you indicated there is the sotck bootloader howver I see many .bin bootloaders and have no clear which one should I use, can you bring me some light?

Thanks in advance

dinamitemic, 2020-01-02:
> Hello Mate,
>
> First of all I wish you a happy new 2020!!!!!!
> Thanks for your add, I see on the link you indicated there is the sotck bootloader howver I see many .bin bootloaders and have no clear [...]

Hi, Happy New Year to you too!! 
You have to download the .zip file posted in the last comment of @thisiskeithb.
It is highlited in the previous link!

davidcgu, 2020-01-02:
thanks a lot!!!!!!!

davidcgu, 2020-01-03:
I order to flash the bootloader or marlin from ST-link it ask me to select the size, can you let me know what I should set on each case?

Thanks in advance

SergeySn, 2020-01-16:
> Like it is mentioned [here](https://github.com/MarlinFirmware/Marlin/pull/14041#issuecomment-513607940) the right addresses in order to program it from stlink are:
> -0x08000000 for the bootloader
> -0x08007000 for the firmware.bin compiled in platformio.

Where should I put the 0x08007000 ?
I see the original posts talk about buildroot/share/platformIO/ldscripts/STM32F103RC_SKR_MINI.py but in my case I already have the correct value there.  Should I change it in some other place as well?

SergeySn, 2020-01-23:
Here is what in my STM32F103RC_SKR_MINI.py
```
import os
Import("env")

STM32_FLASH_SIZE = 256

for define in env['CPPDEFINES']:
    if define[0] == "VECT_TAB_ADDR":
        env['CPPDEFINES'].remove(define)
    if define[0] == "STM32_FLASH_SIZE":
        STM32_FLASH_SIZE = define[1]

# Relocate firmware from 0x08000000 to 0x08007000
env['CPPDEFINES'].append(("VECT_TAB_ADDR", "0x08007000"))

custom_ld_script = os.path.abspath("buildroot/share/PlatformIO/ldscripts/STM32F103RC_SKR_MINI_" + str(STM32_FLASH_SIZE) + "K.ld")
for i, flag in enumerate(env["LINKFLAGS"]):
    if "-Wl,-T" in flag:
        env["LINKFLAGS"][i] = "-Wl,-T" + custom_ld_script
    elif flag == "-T":
        env["LINKFLAGS"][i + 1] = custom_ld_script
```
Does it look okay?

SergeySn, 2020-01-23:
And here is my STM32F103RC_SKR_MINI_256K.ld
```
MEMORY
{
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 48K - 40
  rom (rx)  : ORIGIN = 0x08007000, LENGTH = 256K - 28K
}

/* Provide memory region aliases for common.inc */
REGION_ALIAS("REGION_TEXT", rom);
REGION_ALIAS("REGION_DATA", ram);
REGION_ALIAS("REGION_BSS", ram);
REGION_ALIAS("REGION_RODATA", rom);

/* Let common.inc handle the real work. */
INCLUDE common.inc
```
Are the addresses correct?

labotecno, 2020-05-31:
do i need a st link to repair the board ? 
without is it posible?