bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #9: How to connect a single Z motor?

GilbMata, 2019-06-29:
I need to connect a single motor in Z, but this will not work if it does not have both motors connected, how to operate only one?

ThWoell, 2019-06-30:
Just install a pair of jumper connectors as shown on the frontpage of instructions leaflet. The motors will be connected in serial to the driver, so missing the jumper connectors will leave the motor circuit "open".

GilbMata, 2019-07-02:
> Just install a pair of jumper connectors as shown on the frontpage of instructions leaflet. The motors will be connected in serial to the driver, so missing the jumper connectors will leave the motor circuit "open".

Omg How I did not see those jumpers, I checked it several times

Thank You¡

Rodov, 2021-04-23:
This solution is not obvious, so need to add this into the documentation.