bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #11: Problem with sd-card on reprap smart controller

ThWoell, 2019-07-06:
How can the sd-card reader on the smart controller be used with marlin?
Insertion and removal of card become displayed on display, but when I try to read files from it this won't work. on pronterface there is sd- init error displayed.

3DSmitty, 2019-07-06:
It works with the main Marlin:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

ThWoell, 2019-07-07:
Since i have just adapted my config.h and config-adv.h to work wit the machine is it possible just to copy this files into the marlin source directory?
thanks for helping out again!

ThWoell, 2019-07-08:
Now I used the latest Version of Marlin Bugfix Branch; after minor changes, it worked out; but it is not possible to use the sd-card on the smart controller, but the onboard mini-sd reader works. So at least it is possible to print from card now. any hints on how to switch to the sd-card on the smart controller?

3DSmitty, 2019-07-08:
To use SD enable #define SDSUPPORT and #define SPEAKER 
then in Marlin/src/pins/pins_BIGTREE_SKR_MINI_V1_1.h enable #define STM32_SD_LCD

ThWoell, 2019-07-08:
WORKS!

HGUNNCH, 2019-08-06:
Unfortunately i am having the same issue. If i try to define SPEAKER the control knob no longer works and the sd card continues to no show in the menu, but is showing as being inserted on the LCD. Any help you guys can provide would be great... getting this board to work is driving me insane!

ThWoell, 2019-08-06:
Just follow 3DSmittys advice...

HGUNNCH, 2019-08-06:
Thanks for the quick response. I originally tried with the main Branch but could never get it to compile. The bigtree_mini environment would constantly cause a build error. For some reason the bigtree Firmware compiles just fine though with the the same environment variable.

3DSmitty, 2019-08-06:
Bigtreetech fork SD code will not work correctly their SPI code is wrong. I am the one who wrote the code for the main Marlin 2.0 git repo to get the SD working correctly. They just committed a fix that should allow it to compile easier. Try a fresh checkout.

HGUNNCH, 2019-08-07:
I am currently trying to compile and am getting the following error:

**"In file included from .piolibdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src\source\CHOPCONF.cpp:1:0:
.piolibdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18:11: fatal error: SoftwareSerial.h: No such file or directory"**

I am trying to compile for TMC2208_standalone drivers. I know i am kind of hijacking this thread so if i should start a new one let me know.

ThWoell, 2019-08-09:
@HGUNNCH: First it would be better to open a new issue, second what do you mean with standalone?