bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #12: Weak Power Regulator?

ThWoell, 2019-07-08:
Experimenting with the Board Ii had to find out that the 5V Regulator on the Board is a little weak in Voltage and also sensitive to noisy Input power.
Measured Voltage 4.8V (with 12 V input from adjustable Workbench PSU)on the dc select INT pin
The board will boot most of the time, but sometimes not. When a Hotend fan comes in, wired parallel to the Board, things get worse: Now it will not boot again; just when you hit the reset button it will start the boot sequence.
So if you experience similar problems, please check for the voltage. Since there is no wiring diagram, it is quite a task to find out which regulator on the board will supply what.

NamkingWIN, 2019-12-21:
I have the same problem. Anyone can fix this?

mirokymac, 2020-02-09:
I get my board on Dec.2019. 5V rail is on 4.96V.

Rodov, 2021-04-23:
It work fine for me - the board boots every time. I am using 12 Volt computer PSU.