bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #13: OS can't see the board

100307009, 2019-07-08:
Greetings, 
After trying to install smoothie firmware(compilation successful, loaded firmware.bin into sd card and generic config.txt just to test it), the device stopped being recognized by the OS after that. 
On windows it says that "device failed to provide descriptor" and on linux(mint) pronterface and others are failing to indentify it. 
Any idea what it might be?

3DSmitty, 2019-08-06:
This board does not support smoothieware

anm8tor, 2019-09-11:
I'm having this same problem on my Raspberry Pi 4. Using the main branch of Marlin 2.0 bugfix, the Pi doesn't even register that the USB cable is plugged in. Any thoughts on that? @3DSmitty

Rodov, 2021-03-24:
Can't connect SCR mini 1.1 to Windows 7 x64. It say "Unknown Device". Drivers from GitHub didn't help - it won't install. Updating firmware with Marlin 2.0.7.1 from SD card was successful.