bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #14: I can not connect TMC2208 to the board BIGTREETECH-SKR-MINI-V1.1 with UART

3ona, 2019-07-12:
[image]

`Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\persistent_store_eeprom.cpp.o
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\persistent_store_flash.cpp.o
Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\persistent_store_sdcard.cpp.o
In file included from c:\marlin2\marlin\src\inc\marlinconfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
c:\marlin2\marlin\src\inc\SanityCheck.h:1954:4: error: #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."

   #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from c:\marlin2\marlin\src\inc\marlinconfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
c:\marlin2\marlin\src\inc\SanityCheck.h:1954:4: error: #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from c:\marlin2\marlin\src\inc\marlinconfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp:35:
c:\marlin2\marlin\src\inc\SanityCheck.h:1954:4: error: #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from c:\marlin2\marlin\src\inc\marlinconfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32F1\persistent_store_eeprom.cpp:23:
c:\marlin2\marlin\src\inc\SanityCheck.h:1954:4: error: #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp:36:0:
C:\users\3onap\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:171:15: warning: '_spi3_this' defined but not used [-Wunused-variable]
 static void (*_spi3_this);
               ^~~~~~~~~~
C:\users\3onap\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:168:15: warning: '_spi2_this' defined but not used [-Wunused-variable]
 static void (*_spi2_this);
               ^~~~~~~~~~
C:\users\3onap\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:165:15: warning: '_spi1_this' defined but not used [-Wunused-variable]
 static void (*_spi1_this);
               ^~~~~~~~~~
C:\users\3onap\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:163:16: warning: 'ff' defined but not used [-Wunused-variable]
 static uint8_t ff = 0XFF;
                ^~
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp.o] Error 1
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
In file included from c:\marlin2\marlin\src\inc\marlinconfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32F1\persistent_store_flash.cpp:32:
c:\marlin2\marlin\src\inc\SanityCheck.h:1954:4: error: #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
    ^~~~~
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp.o] Error 1
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\HAL_STM32F1\persistent_store_eeprom.cpp.o] Error 1
In file included from c:\marlin2\marlin\src\inc\marlinconfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32F1\persistent_store_sdcard.cpp:29:`

kristapsdravnieks, 2019-07-23:
You have to define RX and TX pins in /src/pins/stm32/pins_BIGTREE_SKR_MINI_V1_1.h

add something like according to pins You used. 

#if HAS_DRIVER(TMC2208)
  #define X_SERIAL_TX_PIN    P2_06 // "2.6" on LCD/SD/SPI (aka AUX3)
  #define X_SERIAL_RX_PIN    P2_06
  #define Y_SERIAL_TX_PIN    P1_31 // "1.31" on LCD/SD/SPI
  #define Y_SERIAL_RX_PIN    P1_31
  #define Z_SERIAL_TX_PIN    P1_23 // "1.23" on LCD/SD/SPI
  #define Z_SERIAL_RX_PIN    P1_23
  #define E0_SERIAL_TX_PIN   P0_03 // "RX" on AUX1
  #define E0_SERIAL_RX_PIN   P0_03
#endif

Also you get error abount EEPROM, disable it, I don't thing it is implemented.

rhialto56, 2019-07-30:
Pc10, pc11, pc12 can be used.

albmargar30, 2019-10-29:
When you connect the tmc2208 as it is placed on the board, do you have to remove the jumpers underneath or leave them on? In skr 1.3 they are removed but in the manual of the skr mini V1.1 it does not come in the manual as it is installed. I know that pins PC10, PC11 and PC12 correspond to the UART of the mcu.

dinamitemic, 2019-11-07:
Same problem here.
Tried to define software serial pin on for example some spare pin on EXP2 but everytime I get TMC CONNECTION ERROR on display. Also from terminal I get all LOW.
Anyone has find a solution?? Right now I'm using it in STANDALONE mode but I would like to turn UART on in order to get extra functionality

chris3081, 2019-12-30:
@dinamitemic if your getting LOW on the terminal its a wiring config issue.. I got it up and running using the information above, plugging into to PC10, PC11, PC12, and PC14 (for E0).. As the rest of my printer isn't complete yet I've only been able to confirm that the signal is there. I have documented from memory what I did with the aim of doing a PR back to Big Tree to have it included in the doco. Its currently available here https://github.com/chris3081/BIGTREETECH-SKR-MINI-V1.1/blob/master/Wiring%20diagram/SKR%20mini%20V1.1-TMC2208-UART-Mode-guide.pdf 

If there are any errors please let me know.

kondas, 2020-02-16:
@chris3081's solution works wor me too, thanks a lot!

dinamitemic, 2020-02-16:
@chris3081 I've found a solution too. My problem was that I wasn't performing a complete power cycle to the board, that way the driver weren't well initialized. 
So for anyone else: **please make sure to turn off completely the board (USB and external VCC) after firmware mod on the stepper driver.**
I've noticed also that if you power from USB and then turn on the power supply providing VCC it will not work.

jbaiao, 2020-05-05:
@dinamitemic can you share how do you make EXP2 configurations, please?
I have fysetc tmc2208 v1.2, and give me "all LOW" too....
tks!

Rodov, 2021-04-15:
Is it possible to connect TMC2209 by UART and TFT35 in both modes (touch and 12864) at the same time?