bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #17: How to connect Bltouch

1kemanciii, 2019-08-23:
Hi.
skr mini v1.1 bltouch connections and settings how?

rhialto56, 2019-08-27:
Pb5

Stephan35, 2019-09-21:
@1kemanciii you need a pwm pin , so PA1 & PA2 can be use as PWM , you have to change  pin assignment if you use XMAX or YMAX pinout.
`
#define X_MAX_PIN          PA2
#define Y_MAX_PIN          PA1
`

dinamitemic, 2019-11-07:
Hii, if you are still interested or for everyone else having troubles with BLTouch configuration on SKR MINI here are my mods: 
I've added this:
-In **pins_BIGTREE_SKR_MINI_V1_1.h **  (the pin are located on EXP2 connector)
//Servo Pin
#define SERVO0_PIN         PB8
// Z Probe must be this pin
#define Z_MIN_PROBE_PIN    PB9
-In **configuration.h:**
#define BLTOUCH
#define AUTO_BED_LEVELING_BILINEAR
#define Z_SAFE_HOMING

mirokymac, 2020-02-09:
> Hii, if you are still interested or for everyone else having troubles with BLTouch configuration on SKR MINI here are my mods:
> I've added this:
> -In **pins_BIGTREE_SKR_MINI_V1_1.h:** (the pin are located on EXP2 connector)
> //Servo Pin
> #define SERVO0_PIN PB8
> // Z Probe must be this pin
> #define Z_MIN_PROBE_PIN PB9
> -In **configuration.h:**
> #define BLTOUCH
> #define AUTO_BED_LEVELING_BILINEAR
> #define Z_SAFE_HOMING
> **Please note that pins for endstop are not suitable because they have an RC filter on it!!!**

Hi bro, what is the reuslt of this mod?
I got and very poor repeatibility problems with this board, as mention in https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/issues/31

dinamitemic, 2020-02-09:
> Hi bro, what is the reuslt of this mod?
> I got and very poor repeatibility problems with this board, as mention in #31
Can't understand very well what's your question. I've defined that pins in order to do not use the gpio wich has RC filter on it. 
Haven't tested repeatibility.

Dreed420, 2021-07-04:
> Hii, if you are still interested or for everyone else having troubles with BLTouch configuration on SKR MINI here are my mods:
> I've added this:
> -In **pins_BIGTREE_SKR_MINI_V1_1.h:** (the pin are [...]

Hello, i have question
(pins_BT_SKR_mini_V1_1.h(
// BLtouch
#define SERVO0_PIN         PA1  
// Limit Switches
#define Y_MIN_PIN                           PC1
#define X_MIN_PIN                           PC2
#define Z_MIN_PIN                           PC0
//#define Y_MAX_PIN                           PA1
//#define X_MAX_PIN                           PA2
#define Z_MAX_PIN                           PC3 

(configuration.h)
#define Z_MIN_PROBE_PIN PA1 // Pin 32 is the RAMPS default  

its ok ? or im wrong

Marder666, 2022-04-07:
> @1kemanciii you need a pwm pin , so PA1 & PA2 can be use as PWM , you have to change pin assignment if you use XMAX or YMAX pinout. ` #define X_MAX_PIN PA2 #define Y_MAX_PIN PA1 #define Z_MAX_PIN PC3 //#define X_MIN_PIN PC2 //#define Y_MIN_PIN PC1 //#define Z_MIN_PIN PC0
> 
> `
> 
> to
> 
> ` #define X_MAX_PIN PC2
> 
> #define Y_MAX_PIN PC1 #define Z_MAX_PIN PC0 //#define X_MIN_PIN PC2 //#define Y_MIN_PIN PC1 //#define Z_MIN_PIN PC0 `
> 
> And defin bltouch pin to PA1 or PA2
> 
> not possible if you also use XMIN connectors

With this MOD in Marlin... where must be pluged in the Cables?