bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #20: USB not recognized on Raspberry Pi 4

anm8tor, 2019-09-11:
I have Marlin 2.0bugfix firmware installed on a Raspberry Pi 4 , running Octopi 0.17. When I connect the Mini board to the Pi using a USB cable, the Pi doesn't register "seeing" the connection at all. Clearing dmesg after boot, but before connecting, then running dmesg after connecting produces nothing. Checking lsusb only shows the four ports. I can connect the Mini to my PC and communicate via Repetier Host just fine, so I have it narrowed down to a possible linux driver issue on the board, maybe? Or some settings in one of the files used to compile firmware.bin?

OutsourcedGuru, 2019-09-19:
In theory, the two USB 2.0-compatible connectors would be the most like the Raspberry Pi 3B in behavior. I wonder if anyone has tried this with Raspbian Buster on the 3A+.

anm8tor, 2019-09-27:
Anyone able to respond? BigTreeTech???

chris3081, 2020-01-25:
Pi 3b+ is fine for me. I've gone down the route of using the TFT pin 5 connector with TTL converter as it fits better in my Wanhao i3 plus clone.

kondas, 2020-02-16:
I have the same issue, Pi4B + SKR mini 1.1.  Also tried setting over_voltage in rpi boot/config but it did not help. Other devices recognize the board, rpi4 doesnt, dmesg empty. Other usb devices work well with raspberry.

chriskaye, 2024-10-27:
Hi everyone,
I know this is an old thread, but wanted to chip in that I have exactly the same issue with a Raspberry Pi Zero 2W.
BTT mini v1.1 not detected when plugged in to the Pi Zero 2W, but connects instantly on my Pi3b+.
Nothing shown on lsusb or dmesg on the rPi02W.

Did anyone find an answer to this issue?
I flashed an archaic Melzi board with Klipper FW and this connects just fine with the same USB hub and cables on the rPI Zero2W - I can't explain what is happening and have no idea how to troubleshoot this 🤔