bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #31: Very poor repeatibility on BLTOUCH

mirokymac, 2020-02-09:
Hi.

I get very poor BLTOUCH repeatibility on SKR mini v1.1.
The PIN arrangement as shown below:
```
// SERVOS
#define SERVO0_PIN         PA1

//
// Limit Switches
//
#define X_MIN_PIN          PC2
#define X_MAX_PIN          PA2
#define Y_MIN_PIN           PC1
//#define Y_MAX_PIN          PA1
#define Z_MIN_PIN          PC0
#define Z_MAX_PIN          PC3
```
And I am using the TMC2208 in UART mode.

Using the condfigure above, and `#define HOMING_FEEDRATE_Z  (4*60)`, I got a M48 test result of **0.15-0.36**, significantly greater than what I got from the Ender-3 Official board's **0.005-0.001**.

Then I have tried some trouble shooting measures:
1. Ensure BLTOUCH is working as expected - Yes, switch between SKR-MINI and Melzi (Creality) get repeatable result mentioned as above.
2. remove `C7` and replace `R9` with a 0ohm resistence - No help
3. compile with `#define BLTOUCH_FORCE_SW_MODE` - No help
4. remove `C8` and replace `R11` with a 0ohm resistence - No help
5. replace BLTOUCH wire with sheild and twisted wire - still No help
6, add several MLCCs on 5V and GND on J4 - still No help

Any suggestions?

mirokymac, 2020-02-12:
update 2020.2.12:
After change the PIN setup following SKR-mini-E3 using `X_STOP_PIN` instead of `X_MIN_PIN` and etc. making the deviation down to 0.05 result of `M48 P25 V4`.
I have got a -0.135±0.005 deviation at the rate of 10% of probing.
I also check the approach in https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/issues/17, no improve has been inspected.
Still finding way to get around.

mirokymac, 2020-02-14:
update 2020.2.13:
After disabling `#define BLTOUCH_FORCE_SW_MODE` all things going well. this can be closed now.

closed 2020-02-14 by mirokymac

Dreed420, 2021-07-03:
configuration.h

#define Z_MIN_PROBE_PIN PA1 // Pin 32 is the RAMPS default <- its ok ?

mirokymac, 2021-07-06:
> configuration.h
> 
> #define Z_MIN_PROBE_PIN PA1 // Pin 32 is the RAMPS default <- its ok ?

Using #define in <configuration.h> may not be effected. You can try without the stepper connected.

you can find the pin define at the <boardname.h> file. I copied the default BTT one and made changes to it.

Dreed420, 2021-07-11:
> 
> 
> > configuration.h
> > #define Z_MIN_PROBE_PIN PA1 // Pin 32 is the RAMPS default <- its ok ?
> 
> Using #define in <configuration.h> may not be effected. You can try without the stepper connected.
> 
> you can find the pin define at the <boardname.h> file. I copied the default BTT one and made changes to it.

Hi, its my pins_btt_skr_miniv_1_1
[code]// Limit Switches
#define SERVO0_PIN                          PA1
#define X_MIN_PIN                           PC2
#define X_MAX_PIN                           PA2
#define Y_MIN_PIN                           PC1
//#define Y_MAX_PIN                           PA1
#define Z_MIN_PIN                           PC0
#define Z_MAX_PIN                           PC3[/code]

2 pins from bltouch connected to Z- 

configuration
[code]#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN[/code]

And i have problem - bltouch deploy only one time.
When homing, touch bed one time and i have error becouse second time dont push pin out  - any sugestion ? 

sorry for my english