bigtreetech/BIGTREETECH-SKR-MINI-V1.1 #35: Z not working

nobaj, 2021-04-15:
I don't know why my z motor is not moving, if I connect the motor to another axis, it works.
. 
I also replaced the driver (they are both new drivers) but this does nothing. 

Also I move the steps down to 80 to see if it was something to do with the steps per mm. Still nothing. 

I tried both connectors, (I know they are in parallel with the same driver) 

I used the same cable to connect the motor to another axis and it worked, (same cable, same motor different axis) 

Is there a conf or something I need to change on marlin ?

nobaj, 2021-04-15:
I want to rule out everything before saying it is a faulty motherboard

Rodov, 2021-04-23:
This problem was mentioned before and has been successfully solved:
#9 (How to connect a single Z motor?)
Please close the issue, if everything work fine.

closed 2021-04-24 by nobaj